sifive/ProcKami #7: Change the Processor Core so that the Reg Writer Reads CSR Values Before CSR Writes Rather than the Register Reader Task

llee454, 2019-05-08:


llee454, 2019-05-09:
The CSR functional unit (Zicsr.v) requires the CSR value to compute the new CSR value. The new CSR value is computed based on the input register/immediate value, and a variable mask. Originally, I thought that I could simply pass mask that needs to be applied to the register writer, but different instructions apply the mask to different values (register/immediate values). In general to compute the result of a CSR instruction I need three datums: the original CSR value, the mask to apply, and the value to apply the mask to (original csr value, or immediate value).

vmurali, 2019-05-10:
I believe the comment is not valid anymore.

Zicsr must be a special functional unit that is executed only in the final stage. It should be treated as a special function, just like memory.

closed 2019-05-17 by llee454