sifive/ProcKami #16: Make instMisalignedException memMisalignedException and accessException either gallina parameters or part of extension

vmurali, 2019-05-17:
"instMisalignedException?" :: Bool ;
        "memMisalignedException?"  :: Bool ;
        "accessException?"         :: Bool

llee454, 2019-08-08:
derived from device table

llee454, 2019-09-16:
> I added another parameter : misaligned_access

this means, every misaligned exception is converted into an access exception
we can do this transformation in the very end
what is the last step where you take care of exceptions ?
Larry Lee 5:18 PM
trapException in Commit.v
But this has been heavily modified in my debug-modules branch.
So any modifications will definitely generate Git conflicts.
Murali Vijayaraghavan 5:18 PM
OK I will not touch it
but basically, there's this "misaligned_access" bool
new messages
which dictates wheteher Misaligned should be recategorized as Access
So if it's InstMisaligned/SAmoMisaligned/LoadMisaligned, you make them InstAccess/SAmoAccess/LoadAccess respoectively

closed 2019-09-18 by vmurali