sifive/ProcKami #44: pmp failure, device failure, physical address overflow results in access fault

vmurali, 2019-07-31:


llee454, 2019-08-01:
bf18068

llee454, 2019-08-01:
Define a function named `checkForAccessFault : ActionT ty Bool` that performs the pmp, physical address upper bound, and physical device range checks. Then insert a call to checkForAccessFault before every memory read/write operation in the page table walker, mem_unit_exec, and memFetch; and raise an access exception if this function returns true, which indicates that an access exception occured.

vmurali, 2019-08-01:
The function must return 
`LetExprSyntax ty (Maybe (Bit (lg num devices))`

llee454, 2019-08-01:
Represent device tags as Bit (log2_up num_devices).

llee454, 2019-08-02:
1.Add PMA to MemDevice record type
make read functions accept a (device offset : PAddr @# ty) and return Maybe Data.
make write functions accept a (device offset : PAddr @# ty) and return Bool.
add permissions (read, write, execute) check action.
add supportsAmo : ActionT _ Bool
add supportsLrSc : ActionT _ Bool
add supportsUnaligned - for unaligned reads and writes
add readGranularity
add writeGranularity
2. Add optional register file or memory mapped register to MemDevice record type
will be used to generate registers and register files from device table
3. Change mem_devices table from a vector to a list of MemDevice and make device indicies be Fin.t (length mem_devices).
Apparently, the Haskell simulator performs vector operations slowly.
4. Change the way in which checkForAccessException computes the device address (which we now call device offset).
(paddr - (mem_region_start curr_mem_region)) + (list_sum (map mem_region_size (filter (mem_region_has_device (mem_region_device curr_mem_region)) prev_mem_regions)))
5. Change the type of the memory regions table to ((start : nat) * list MemRegion), where memRegion := {width : nat; mdevice : option MemDevice}.
This change prevents overlapping regions, and makes offset calculations easy.
Used internally
7. Define function to map from current memory regions table type to new memory regions table type
current type is easy to enter but may contain overlaps.
Make sure this function returns an maybe type if input value is invalid.
8. Define Lemma asserting that the input table is valid - i.e. mapping function is not None.

llee454, 2019-08-12:
c4c59b4

closed 2019-08-12 by llee454