sifive/ProcKami #95: Profile linux run

vmurali, 2019-09-04:


llee454, 2019-09-04:
Use trial memory constraints and set the timeout to 1000 instructions to generate time estimates.

llee454, 2019-09-04:
Use Haskell memory profiling to measure the memory consumption of the Linux processor.

llee454, 2019-09-06:
10 instructions in linux with profiling enabled and PMemDevice size at 2^24:
- `( srun --cpus-per-task=1 --mem=20G ./runElf.sh --path rv64-vmlinux --haskell --xlen 64 ) &> linux64smallmemprofiled.output`

```
real    3m17.418s
user    2m42.284s
sys     0m34.904s
```
Note: the trace was missing almost all output.
20 instructions. There appears to be a ~3 minute initialization delay followed by reasonably fast execution 
```
real    3m35.461s
user    2m50.088s
sys     0m44.980s
```
100 instructions
```
real    4m38.597s
user    4m33.932s
sys     0m4.108s
```
```
1             Fri Sep  6 12:04 2019 Time and Allocation Profiling Report  (Final)
    1 
    2            Main +RTS -p -h -RTS boot_rom=boot_ROM_RV64.hex testfile=haskelldump/rv64-vmlinux.hex xlen@64
    3 
    4         total time  =       22.03 secs   (22032 ticks @ 1000 us, 1 processor)
    5         total alloc = 53,756,816,256 bytes  (excludes profiling overheads)
    6 
    7 COST CENTRE          MODULE             SRC                                        %time %alloc
    8 
    9 adjust               Data.HashMap       Data/HashMap.hs:(387,1)-(391,52)            58.0   64.0
   10 parseHex             Simulator.Parse    Kami/Simulator/Parse.hs:(57,1)-(66,51)      12.1    8.2
   11 parseHex.v_init      Simulator.Parse    Kami/Simulator/Parse.hs:61:13-51            11.9    8.0
   12 getToks              Simulator.Parse    Kami/Simulator/Parse.hs:29:1-130             2.8    5.5
   13 sysIO                Simulator.Print    Kami/Simulator/Print.hs:(51,1)-(63,82)       2.3    0.7
   14 parseHex.val_pairs   Simulator.Parse    Kami/Simulator/Parse.hs:63:13-78             2.1    3.0
   15 toks_to_addr_vals.go Simulator.Parse    Kami/Simulator/Parse.hs:(39,5)-(41,51)       1.9    2.1
   16 eval                 Simulator.Evaluate Kami/Simulator/Evaluate.hs:(70,5)-(93,94)    1.1    0.3
   17 readTok.txt'         Simulator.Parse    Kami/Simulator/Parse.hs:22:21-48             1.1    2.5
   18 readTok              Simulator.Parse    Kami/Simulator/Parse.hs:(22,1)-(26,56)       1.0    0.8
```
1000 instructions
```
real	31m10.743s
user	31m0.568s
sys	0m7.940s
```
```
1             Fri Sep  6 12:11 2019 Time and Allocation Profiling Report  (Final)
    1 
    2            Main +RTS -p -h -RTS boot_rom=boot_ROM_RV64.hex testfile=haskelldump/rv64-vmlinux.hex xlen@64
    3 
    4         total time  =      151.35 secs   (151354 ticks @ 1000 us, 1 processor)
    5         total alloc = 377,311,253,760 bytes  (excludes profiling overheads)
    6 
    7 COST CENTRE     MODULE             SRC                                        %time %alloc
    8 
    9 adjust          Data.HashMap       Data/HashMap.hs:(387,1)-(391,52)            80.8   87.7
   10 sysIO           Simulator.Print    Kami/Simulator/Print.hs:(51,1)-(63,82)       3.6    1.0
   11 eval            Simulator.Evaluate Kami/Simulator/Evaluate.hs:(70,5)-(93,94)    2.0    0.4
   12 parseHex        Simulator.Parse    Kami/Simulator/Parse.hs:(57,1)-(66,51)       1.7    1.2
   13 parseHex.v_init Simulator.Parse    Kami/Simulator/Parse.hs:61:13-51             1.5    1.1
```

closed 2019-09-06 by llee454