sifive/ProcKami #96: Check --noprint performance without profiling

vmurali, 2019-09-04:


llee454, 2019-09-04:
5891 instructions in 56m5.100s with printing statements run locally on gamma03

llee454, 2019-09-04:
larryl@gamma03:/nettmp/netapp1a/llee/RiscvSpecFormal-090419$ time ./Main boot_rom=boot_ROM_RV64.hex testfile=haskelldump/rv64-vmlinux.hex xlen@64 --noprint > haskelldump/rv64-vmlinux-090419b.out
^[[BKilled

real	61m34.773s
user	6m52.384s
sys	31m37.908s

llee454, 2019-09-04:
larryl@gamma03:/nettmp/netapp1a/llee/RiscvSpecFormal$ time ./runElf.sh --xlen=64 --path riscv-tests/rv64ui-p-add --haskell --noprint
Notice: Running rv64ui-p-add
Passed

real	0m9.536s
user	0m9.312s
sys	0m0.144s

real	0m9.687s
user	0m9.320s
sys	0m0.144s
larryl@gamma03:/nettmp/netapp1a/llee/RiscvSpecFormal$ time ./runElf.sh --xlen=64 --path riscv-tests/rv64ui-p-add --haskell
Notice: Running rv64ui-p-add
Passed

real	0m13.799s
user	0m11.496s
sys	0m2.284s

real	0m13.844s
user	0m11.504s
sys	0m2.284s

llee454, 2019-09-05:
without the noprint flag, the simulator took this long to execute 500 instructions from the linux binary:
```
real    7m38.117s
user    7m8.876s
sys     0m29.244s
```

llee454, 2019-09-05:
with the noprint flag, the simulator took this long to execute 500 instructions from the linux binary:
```
real    7m54.120s
user    7m7.680s
sys     0m46.356s
```

llee454, 2019-09-05:
32 bits 64 bits tests
profiling on profiling off
time
print no print

llee454, 2019-09-05:
The following tests will be run:

llee454, 2019-09-05:
Aborted all remaining jobs in phase 1 after all ran for more than an hour and one became unresponsive:
```
             JOBID PARTITION     NAME     USER ST       TIME  NODES NODELIST(REASON)
          45203885  standard     time   larryl  R    1:03:18      1 gamma13
          45203884  standard     time   larryl  R    1:03:38      1 gamma13
          45203881  standard     time   larryl  R    1:03:54      1 sigma12
          45203778  standard     time   larryl  R    1:04:09      1 sigma12
```

llee454, 2019-09-05:
Running runTests to verify that the binaries can run any tests successfully.

llee454, 2019-09-05:
test 1:  (srun --cpus-per-task=1 --mem=8G ./runElf.sh --path /nettmp/netapp1a/vmurali/riscv-tests/isa/rv64ui-v-add --haskell --xlen 64) &> add64.output &
test 2: (srun --cpus-per-task=1 --mem=8G ./runElf.sh --path /nettmp/netapp1a/vmurali/riscv-tests/isa/rv32ui-v-add --haskell --xlen 32) &> add32.output &
test 3: (srun --cpus-per-task=1 --mem=8G ./runElf.sh --path /nettmp/netapp1a/vmurali/riscv-tests/isa/rv64ui-v-add --haskell --xlen 64 --noprint) &> add64quiet.output &
test 4: (srun --cpus-per-task=1 --mem=8G ./runElf.sh --path /nettmp/netapp1a/vmurali/riscv-tests/isa/rv32ui-v-add --haskell --xlen 32 --noprint) &> add32quiet.output &

llee454, 2019-09-05:
run linux for 8000 instructions without the noprint flag.

llee454, 2019-09-06:
Linux executed 8000 instructions with the following time stats with the print flag enabled:
```
real    77m25.536s
user    70m49.212s
sys     6m35.948s
```

llee454, 2019-09-06:
Linux execute 500 instructions with the UART device removed from the memory table and memory devices in ModelParams.v

llee454, 2019-09-06:
Linux execute 500 instructions normal config:
```
real    7m19.149s
user    6m52.016s
sys     0m27.080s
```

llee454, 2019-09-06:
Comparison run for rv64-v-add with the UART device removed from the memory regions table and memory devices table in ModelParams.v.
```
real    8m25.505s
user    8m22.024s
sys     0m3.336s
```

llee454, 2019-09-06:
Comparison run for rv64-v-add 500 instructions, 2^20 mem, uart enabled.
- `(srun --cpus-per-task=1 --mem=20G ./runElf.sh --path /nettmp/netapp1a/vmurali/riscv-tests/isa/rv64ui-v-add --haskell --xlen 64) &> add64.output3 &`

llee454, 2019-09-06:
Linux 500 instructions with 2^24 mem, uart enabled
* `(srun --cpus-per-task=1 --mem=20G ./runElf.sh --path rv64-vmlinux --haskell --xlen 64) &> linux64smallmem.output`
```

llee454, 2019-09-06:
10 instructions in linux with profiling enabled and PMemDevice size at 2^24:
- `( srun --cpus-per-task=1 --mem=20G ./runElf.sh --path rv64-vmlinux --haskell --xlen 64 ) &> linux64smallmemprofiled.output`

```
real    3m17.418s
user    2m42.284s
sys     0m34.904s
```
Note: the trace was missing almost all output.

sifive-emarzion, 2019-09-07:
I tested making the following change to `sysIO` (printing an empty string instead of returning `()`:

```
sysIO :: Modes -> T.SysT Val -> IO ()
sysIO modes T.Finish = do
    let no_print = no_print_mode modes
    let interactive = interactive_mode modes
    if (not no_print && not interactive) then hPutStrLn stdout "Exiting..." else hPutStrLn stdout ""
    exitSuccess
sysIO modes (T.DispString msg) = do
    let no_print = no_print_mode modes
    let interactive = interactive_mode modes
    if (not no_print && not interactive) then hPutStr stdout $ format_string $ msg else hPutStrLn stdout ""
sysIO modes (T.DispExpr _ e ff) = do
    let no_print = no_print_mode modes
    let interactive = interactive_mode modes
    if (not no_print && not interactive) then hPutStr stdout $ printVal ff $ eval e else hPutStrLn stdout ""
```

Here are the corresponding profiling reports with or without `--noprint` (test is rv64ui-p-add and using memory size of 2^20):

```
        Fri Sep  6 17:31 2019 Time and Allocation Profiling Report  (Final)

           Main +RTS -p -RTS boot_rom=boot_ROM_RV64.hex --noprint testfile=haskelldump/rv64ui-v-add.hex tohost_address:1000 xlen@64

        total time  =      142.46 secs   (142456 ticks @ 1000 us, 1 processor)
        total alloc = 196,032,161,360 bytes  (excludes profiling overheads)

COST CENTRE                                                                                                 MODULE              SRC                                         %time %alloc

eval                                                                                                        Simulator.Evaluate  Kami/Simulator/Evaluate.hs:(70,5)-(93,94)    26.0    6.3
simulate_action.sim                                                                                         Simulator.Simulator Kami/Simulator/Simulator.hs:(47,5)-(84,74)    7.1   10.3
eval                                                                                                        Simulator.Evaluate  Kami/Simulator/Evaluate.hs:(31,5)-(35,92)     3.7    2.5
adjust                                                                                                      Data.HashMap        Data/HashMap.hs:(387,1)-(391,52)              3.0    3.6
convertLetExprSyntax_ActionT                                                                                Syntax              Haskell/Syntax.hs:(531,1)-(539,52)            2.9    6.2
sysIO                                                                                                       Simulator.Print     Kami/Simulator/Print.hs:(51,1)-(63,108)       2.9    1.1
div_rem_full.\                                                                                              Div                 Haskell/Div.hs:217:57-65                      2.8    6.1
convertLetExprSyntax_ActionT.\                                                                              Syntax              Haskell/Syntax.hs:536:12-50                   2.4    0.0
eval                                                                                                        Simulator.Evaluate  Kami/Simulator/Evaluate.hs:(46,5)-(51,63)     1.9    1.1
func_units                                                                                                  ModelParams         Haskell/ModelParams.hs:(136,1)-(236,49)       1.8    0.3
coq_FDivSqrt.\.\                                                                                            FDivSqrt            Haskell/FDivSqrt.hs:(286,5)-(289,66)          1.7    5.0
envPre                                                                                                      Main                Main.hs:(94,3)-(162,24)                       1.5    0.7
nth_Fin.\                                                                                                   EclecticLib         Haskell/EclecticLib.hs:25:25-39               1.5    0.0
coq_Mac.\.\                                                                                                 FMac                Haskell/FMac.hs:(587,66)-(623,76)             0.8    1.8
coq_MulAdd_expr.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\ MulAdd              Haskell/MulAdd.hs:(2885,27)-(3305,34)         0.7    2.0
utila_expr_lookup_table                                                                                     Utila               Haskell/Utila.hs:(246,1)-(252,75)             0.5    1.1
coq_INToNF_expr.\.\.\.\.\.\.\                                                                               INToNF              Haskell/INToNF.hs:(228,124)-(247,32)          0.4    1.1
utila_opt_pkt                                                                                               Utila               Haskell/Utila.hs:(39,1)-(40,103)              0.3    1.1
decode_match_field                                                                                          Decoder             Haskell/Decoder.hs:(136,1)-(151,64)           0.2    1.1
loopFn                                                                                                      Div                 Haskell/Div.hs:(44,1)-(211,49)                0.1    1.0
loopFn.\.\                                                                                                  Div                 Haskell/Div.hs:(68,39)-(211,47)               0.1    1.0
[... cut ...]
coq_Mac.\.\                                                                                                 FMac                Haskell/FMac.hs:(587,66)-(623,76)             0.6    1.4
coq_MulAdd_expr.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\.\ MulAdd              Haskell/MulAdd.hs:(2885,27)-(3305,34)         0.5    1.6
```
So using this hacky solution, `--noprints` no longer crashes and actually improves time and space usage as expected (though perhaps not as much as if the print was removed entirely).  It then must be the case that something wonky is going on with Haskell IO.

vmurali, 2019-09-07:
I think we should reimplement the polling mechanism to avoid this memory
leak in Haskell instead of using the hack

closed 2019-10-14 by llee454