sifive/ProcKami #98: Add generic checkpointing for simulation

vmurali, 2019-09-04:


llee454, 2020-02-28:
- [ ] port interactive feature
- [ ] after talking with Evan, add state save and restore feature

vmurali, 2020-03-23:
```
initialize: KamiModule -> KamiState
simulate: Meth1 -> Meth2 -> ... Methn -> KamiState -> KamiState
createCheckpoint: KamiState -> String -> IO()
readCheckpoint: String -> IO KamiState
getRegVal: KamiState -> String -> Value
getRegFileValInt: KamiState -> String -> Int -> Value
putRegVal: KamiState -> String -> Value -> KamiState
putRegFileValInt: KamiState -> String -> Int -> Value -> KamiState
where 
  KamiState = (Map String Value * FileState)
```

vmurali, 2020-03-25:
Repeat TIMEOUT and FAILED so that it is more noticable.

closed 2020-04-03 by sifive-emarzion