sifive/ProcKami #104: Fix the WFI implementation to not be a NOP

vmurali, 2019-09-18:


vmurali, 2020-03-06:
If WFI is executed, go into a (new WFI) state which does not execute any rule, except for checking pending interrupts.

When an interrupt is pending, that state is cleared and normal operation is resumed.

llee454, 2020-03-06:
- [ ] Create "isWfi : Bool" register
- [ ] Add "wfiTag" to the execute context update packet
- [ ] Modify the commit unit to set the isWfi register
- [ ] Modify the pipeline rules so that they do not execute if "isWfi" is true
- [ ] Modify the trapInterrupt rule so that it sets "isWfi" to false.

llee454, 2020-03-06:
> Murali Vijayaraghavan  5:22 PM
> Sorry
> 5:22
> 
>     what is this:  Add "wfiTag" to the execute context update packet ?
> 
> 5:23
> I agree with the rest
> Larry Lee  5:23 PM
> I.e. when the executor executes a wfi tag, it will set val1's tag to WfiTag.
> 5:23
> To signal that we have to wait for an interrupt.
> Murali Vijayaraghavan  5:23 PM
> ok

llee454, 2020-03-09:
b1bea49f9597f09bd6671804bbb1c580f5650e30

closed 2020-03-09 by llee454