sifive/ProcKami #247: Revise the Functional Units

llee454, 2020-02-27:
Make changes to the newone branch and test using the master branch. **Do not commit to master.** 

> Murali Vijayaraghavan  1:10 PM
> InputXform: ExecContextPktWithException -> FunctionalUnitInput
> OutputXform: FunctionalUnitOutput -> ExecUpdatePktWithException
> new messages
> 1:11
> Neither of them should be ID
> 
> Larry Lee  1:11 PM
> Understood.
> 1:11
> That matches my meeting notes. (edited) 
> 
> Murali Vijayaraghavan  1:11 PM
> The functional unit should not, unless you discuss with me and confirm, ever use Exception
> 
> Larry Lee  1:11 PM
> Ok.

llee454, 2020-02-27:
- [ ] Verify that the input transforms never read incoming Exceptions.

llee454, 2020-02-27:
> So, the outputxform should translate that function unit's local exception into a global exception

vmurali, 2020-02-27:
And the final comment is that any commits to ProcKami should happen in the newone branch but tested in master (*not* committed in master) till it gets merged

closed 2020-03-03 by llee454