sifive/ProcKami #297: TileLink Mask

vmurali, 2020-03-25:
```
64: 000 -> 1111_1111
32: 000 -> 0000_1111
32: 100 -> 1111_0000
16: x00 -> 0000_0011
16: x01 -> 0000_1100
16: x10 -> 0011_0000
16: x11 -> 1100_0000
 8: 000 -> 0000_0001
 8: 001 -> 0000_0010
 8: 010 -> 0000_0100
 8: 011 -> 0000_1000
 8: 100 -> 0001_0000
 8: 101 -> 0010_0000
 8: 110 -> 0100_0000
 8: 111 -> 1000_0000
```

llee454, 2020-03-25:
[ where | size ]

llee454, 2020-03-25:
- [ ] Compute the mask function on the examples given above.

llee454, 2020-03-26:
https://github.com/sifive/ProcKami/pull/300

llee454, 2020-03-26:
- [ ] Move mask and shift functions to MemOpsFuncs.v
- [ ] Merge tests and refactored mask and shift functions with current master.

llee454, 2020-03-26:
- [ ] revise shiftAmt in TlUH.v so that it accepts a PAddr rather than a nat.
- [ ] revise the tests for shiftAmt so that the upper bits of the PAddr are random while the lower 3 bits cover all of the cases illustrated in the table above.
- [ ] convert the compute statements in the tests for shiftAmt and mask to Goal statements.
- [ ] add hex equivalents to the table given above in this Issue.

llee454, 2020-03-26:
https://github.com/sifive/ProcKami/pull/301

closed 2020-03-26 by llee454