sifive/ProcKami #319: Trigger for debug

vmurali, 2020-04-03:


llee454, 2020-04-07:
- [ ] Draft function that accepts trigger module parameters and generates the underlying registers
- [ ] Draft functions that accept the trigger module data structs and maps these onto underlying register writes
- [ ] Draft functions that accept the underlying register values and returns the trigger module data structs

llee454, 2020-04-07:
https://sifive.atlassian.net/wiki/spaces/CIP/pages/331513894/SiFive+Debug+System

vmurali, 2020-04-14:
Possible Abstract Structure:

Record AbsStruct (A: Type) {
   RegKind: Kind;
   AbsKind: Kind;
   fromAbs: AbsKind @# ty -> RegKind @# ty;
   toAbs: RegKind @# ty -> AbsKind @# ty;
}

llee454, 2020-04-16:
- [ ] renamed numTriggers to lgNumTrigs

llee454, 2020-04-22:
- [ ] place aliases for StructPkt _ and StructRegPkt _ in FU.v